Zendure/developer-device-data-report #2: Help to install into Fhem

tasisale, 2023-04-26:
Can you give a example how to add your code to Fhem?

Zendure, 2023-04-27:
To integrate Fhem with MQTT and subscribe to messages starting with a user's appKey, you can follow these steps:

Install the MQTT plugin in Fhem. You can use the following commands to install the plugin:

update all
install MQTT
Configure the MQTT connection. Add the following content to the Fhem configuration file:

define mqtt MQTT <broker-address>:<port>
attr mqtt user <username>
attr mqtt password <password>
Where <broker-address> is the address of the MQTT broker server, <port> is the port number, <username> and <password> are the username and password required to connect to the MQTT broker server.

Subscribe to messages starting with the user's appKey. Add the following command to the Fhem configuration file:

define my_mqtt_sub MQTT_SUB <topic-filter>
Where <topic-filter> is the MQTT topic filter you want to subscribe to, such as {appKey}/#.

After completing these steps, Fhem will be able to connect to the MQTT broker server and subscribe to all MQTT messages starting with appKey. Please note that this assumes you have correctly configured your MQTT client and server and that your Fhem system is able to communicate with the MQTT broker server.

tasisale, 2023-06-05:
> @tasisale To get your app key, follow the readme instructions here: https://github.com/Zendure/developer-device-data-report I used Postman to post the message and get the information

ok, but why?
sorry for the bad question.
how i will connect it finaly to fhem?

thx for yout help.

tasisale, 2023-07-05:
no one who can help?

steffenxx, 2023-10-12:
i use NodeRed to read the MQTT Values from the Zendure MQTT Server. Then NodeRed writes some Value to influxDB/Grafana and some Values i write via MQTT direkt in FHEM Device readings. sounds complicated but it is not.

steffenxx, 2023-10-12:
thx, it works :-)