Zendure/developer-device-data-report #4: only 2 of 3 items in the feed

copadewe, 2023-05-06:
Hi zendure,
as long as we use "try and error" there will come several questions.
Please publish a working integration.

For me, there are only 2 of my 3 items sending feeds:
[image]

Can you explain what to do? And is this a SBV and the connected Battery, OR are this my 2 SBVs?
Oliver

Zendure, 2023-05-06:
Hi Oliver,
Based on your feedback, we understand that you are experiencing some issues with integrating your power bank and only two out of three items are sending feeds. We suggest checking if the firmware for your device is up-to-date and if it is correctly connected to your network.

copadewe, 2023-05-06:
> Hallo Oliver, Basierend auf Ihrem Feedback verstehen wir, dass Sie einige Probleme mit der Integration Ihrer Powerbank haben und nur zwei von drei Artikeln senden Feeds. Wir schlagen vor, zu überprüfen, ob die Firmware für Ihr Gerät aktuell ist und ob sie korrekt mit Ihrem Netzwerk verbunden ist.

Hello, all three items are on, connected and visible in the app.
and the last firmware ist also installed.

I have like in the other issue, problems to bring it into HA. Because the broker only allow 1 mqtt connection,
and we have used it already. So we try to setup f.e. Mosquitto  Broker as bridge like this:

connection external-bridge
address mqtt.zen-iot.com:1883
topic V7k86PBe/# both 2
remote_username xxxxxxx
remote_password xxxxxxxx
clientid = same as remote_username

But it looks like it´s not bringing details in.
regards
Oliver

Zendure, 2023-05-06:
hello there，We plan to release a new firmware update for Satellite Battery B4600 next Monday to address this issue, so please check for updates then. In the meantime, you could try restarting these devices and ensuring that they are properly connected to your network.

copadewe, 2023-05-06:
> hello there，We plan to release a new firmware update for Satellite Battery B4600 next Monday to address this issue, so please check for updates then. In the meantime, you could try restarting these devices and ensuring that they are properly connected to your network.

Ok thank you ! But the main question is, how we get the information into HA.
will you do a repostity that we can easiely integrate than? than we don´t have to 
work on experimental things :-)

lbouriez, 2023-05-18:
Check that: https://github.com/Zendure/developer-device-data-report/issues/3