Zendure/developer-device-data-report #5: acOutputVoltage wrong data

sirnicolas21, 2023-05-11:
acOutputVoltage for me should be 230V since I am in Europe/Greece
but its 120V

[image]

Zendure, 2023-05-15:
Thank you very much for your feedback. Based on the information you provided, I can tell you that the device uses a dual hot-wire output, and the 230V voltage in Europe is obtained by adding two out-of-phase 115V voltages together. Therefore, the reported voltage of a single hot wire is approximately 115V. If you have any further questions or need further assistance, please feel free to contact me anytime.

sirnicolas21, 2023-05-15:
and you cannot see inside the API code that the mode is 230v and make the variable correct? its phaseA + phaseB ...... or are you saying that the API will give erroneous data all the time?

lbouriez, 2023-05-18:
@Zendure It seems strange that the sensor give a 110v output when in reality it's 220v, I am not an electrician and not very interested to know the technical details, just wish to get the proper value here.

Zendure, 2023-05-22:
The device is divided into two phases, A and B, internally. The actual working voltage is the sum of the two phase voltages. We are currently considering whether to display the total voltage or report it according to a single phase. Please stay tuned for updates.

sirnicolas21, 2023-06-21:
resolved

closed 2023-06-21 by sirnicolas21