Zendure/developer-device-data-report #11: missing attributes

sirnicolas21, 2023-06-05:
home assistant is complaining about missing attributes in every mqtt message received by SBV, i have a log of 500gb with that
[image]

after digging I found this
[image]
https://community.home-assistant.io/t/mqtt-dict-object-has-no-attribute/297330/18

sirnicolas21, 2023-06-06:
that works yes, I still however believe that zendure API should behave as HA describes, so I will keep the issue open

sirnicolas21, 2023-06-06:
> I have added the following to Home Assistant configuration.yaml to prevent these log messages:
> 
> ```yaml
> logger:
>   default: info
>   filters:
>     homeassistant.helpers.template:
>       - "Template variable warning: 'dict object' has no attribute"
> ```

that works yes, I still however believe that zendure API should behave as HA describes, so I will keep the issue open

z-master42, 2023-06-29:
> Is there a work around to turn off this logging?

Add a default value to the value_json like ` value_template: "{{ value_json.electricLevel | int('') }}"` or ` value_template: "{{ value_json.electricLevel | default('') }}"`.

AzraelsDisk, 2023-07-20:
I would say there are a lot more of them... Can somebody help me out?

Logger: homeassistant.helpers.template
Source: helpers/template.py:735
First occurred: 14:22:02 (214060 occurrences)
Last logged: 21:57:08

- Template variable warning: 'dict object' has no attribute 'remainInputTime' when rendering '{{ value_json.remainInputTime}}'
- Template variable warning: 'dict object' has no attribute 'packState' when rendering '{{ value_json.packState}}'
- Template variable warning: 'dict object' has no attribute 'packNum' when rendering '{{ value_json.packNum}}'
- Template variable warning: 'dict object' has no attribute 'electricLevel' when rendering '{{ value_json.electricLevel}}'
- Template variable warning: 'dict object' has no attribute 'socSet' when rendering '{{ value_json.socSet}}'

z-master42, 2023-07-20:
See my previous post.

AzraelsDisk, 2023-07-20:
> See my previous post.

yes, but where to add... sorry for the noob question

z-master42, 2023-07-21:
Were your entities created automatically by HA or did you enter them manually in configuration.yaml or similar?

AzraelsDisk, 2023-07-21:
They where created automatically by HA, so no manually edit by my side