Zendure/developer-device-data-report #16: MQTT Explorer

JasonMM2000, 2023-06-28:
Hello,

I have followed the directions to the best of my ability.  I cannot get any response back from the MQTT or MQTT Explorer add-ins.
First I got the username and passwords using the procedure. 
Then I installed Mosquitto Broker in Add-ons
Then I installed Mosquito in Integrations
Then I installed MQTT in Integrations

When configuring MQTT and MQTT Explorer it looks like a good connection / authentication.  It just doesn't send anything back to these plugins.
[image]
[image]

Should I see something once I provide my username and password to MQTT?

Zendure, 2023-06-29:
You need to enable 'Enable discovery' in the MQTT integration configuration and set the 'Discovery prefix' with your appKey
[image]

JasonMM2000, 2023-06-29:
Thank you for the help.  I do now see the field that I missed and I have put my key in there.  I still am not getting any entities found.  Is there a log I can look at to see any responses it is getting?   

Here is my MQTT config:  

[image]

JasonMM2000, 2023-06-29:
Thank you for the help. I do now see the field that I missed and I have put my key in there.

I have seen my first superbase V.  Now to figure out how to get the second one to show up...

Zendure, 2023-06-30:
The same developer account can subscribe to all the devices under that account.

tasisale, 2023-07-05:
> Thank you for the help. I do now see the field that I missed and I have put my key in there.
> 
> I have seen my first superbase V. Now to figure out how to get the second one to show up...

where you get your key ?
i don't find it.
please help!

z-master42, 2023-07-08:
On Linux, open a terminal and type the following command (on one line, of course):
`curl -i -X POST -H 'Content-Type: application/json' -d '{"snNumber": "YourHubSerialNumber", "account": "Zendure-account-email-address"}' https://app.zendure.tech/v2/developer/api/apply`

Under Windows in the command prompt, the command looks like this:
`curl -i -v --json "{'snNumber': 'YourHubSerialNumber', 'account': 'Zendure-Account-EmailAddress'}" https://app.zendure.tech/v2/developer/api/apply`

The response you receive should look like this:
`{"code":200, "success":true, "data":{"appKey": "YourAppKey", "secret": "YourAppSecret", "mqttUrl": "mqtt.zen-iot.com", "port":1883}, "msg": "Successful operation"}`.

You will then find your individual answers at `appKey` and `secret`.

tasisale, 2023-07-11:
hi, thx. i have no the need data.
my connection is now ok, to teh mqtt, but how i get any data?
i will use MQTTX

z-master42, 2023-07-11:
What exactly do you mean? Did you follow the steps of @JasonMM2000 ? What is MQTTX?

tasisale, 2023-07-11:
> What exactly do you mean? Did you follow the steps of @JasonMM2000 ? What is MQTTX?

MQTTX is a Windows Tool to connect and test MQTT-Connections.
I have now add this connection to this tool, but no values / variable i get back to show.
at tow i have add the MQTT_Client into my fhem, the same, connection OK, but no attributes get autocreate.

regards simon