Zendure/developer-device-data-report #17: Smart Plug

neotigger, 2023-06-30:
Hello
I know myself, you guys are busy, I still ask, will there be smart plug support via MQTT?
Not to control the same, but to read the energy consumption (in Home Assistant) for now.

Controlling the plug via MQTT also via Home Assistant, also from Solarflow directly and in combination, would of course be awesome in the long run.

Respect for your great work

Zendure, 2023-06-30:
This is a great suggestion, and we will carefully consider and make efforts to support this issue. If you have any further information or requests, please feel free to let us know, and we will do our best to assist you. Once again, thank you for your feedback and support!

Hraph, 2023-12-26:
Up, this would be an excellent solution, as smart plugs are also sold in Zendure solar panel packages (flexible balcony). This upgrade would not require so much development from your team.

Corentz, 2024-02-19:
Up, we need Smart Plug reading from Home Assistant. Thanks

neotigger, 2024-02-19:
I now use the Smartplug independently of Zendure. I've integrated it into Home Assistant via the Tuya/Smartlife app as a normal adapter plug. It's not officially supported, but it works.

Corentz, 2024-02-19:
Thanks for your reply @neotigger but I use the smartplug to control the solar flow, so I need them in the Zendure's App.
Otherwise I would have purchase much cheaper plugs :)