Zendure/developer-device-data-report #19: Solarflow set outputLimit from Homeassistant / MQTT

nfocke, 2023-07-05:
Thanks for providing the Homeassistant / MQTT integration. it is a bit fuzzy to implement, but doable with the provided instructions.
However, the most crucial missing feature would be to be able to set the "outputLimit" to a new value. If needed / not possible otherwise in the usual 100W steps, smaller steps / free values would be even better.

I tried sending a "outputLimit" MQTT command via the MQTT config page but this seems not to work. I understand this is not (yet) implemented? A smart(er) management of the discharge to home level using the rich set of Homeassistant tools would be very relevant.
Can you give a timeline for this feature? Or is it available but not documented?

usekevin1988, 2023-07-07:
i think alot of people waiting for this. Including me. i dont get it why full MQTT isnt available out of the box. Shouldnt be a Problem

reinhard-brandstaedter, 2023-07-07:
Thumbsup! I just got my SF today and I'm really looking for (a) direct HA integration via MQTT (my own LAN MQTT broker) and (b) the option to set the outputLimit depending on HA logic I've already implemented i.e. setting it to exactly what is currently needed in the house reported by my integrated smartmeter.

z-master42, 2023-07-08:
What kind of inverter do you use?

nfocke, 2023-07-08:
It is a Bosswerk MI600 = Deye Sun 600. SolarFlow is connected to one MPTT input channel, the other channel is connected with a 400 w panel. Hence, the output limit of SolarFlow is set to 300 Watt. The inverter also limits the input to 300 W
 per channel.

view it on GitHub, or 
unsubscribe.

z-master42, 2023-07-08:
Can the inverter be controlled via Home Assistant? That would at least be a possibility as long as the SF is not controllable via MQTT. I do this with my Hoymiles.

nfocke, 2023-07-08:
I see. Have not tried that, but would need to control the input channel directly, since only one is connected to the solarflow and the other has a separate PV module. Total output power is controllable via their app, maybe also via MQTT. Not sure. Thanks for this suggestion. Still, controlling the Solarflow directly would be more straightforward... ;)

reinhard-brandstaedter, 2023-07-08:
@z-master42 I don't think that controlling the limit of the inverter (I did that in the past) would really help here? Wouldn't SF still output the max configured limit and therefore waste the capacity. If SF puts out 400W and the inverter limits it's output to 200...the 200W from SF are lost...

z-master42, 2023-07-08:
I am currently doing exactly that with my Hoymiles. The SF automatically takes the value it gets rid of at the output, i.e. what the inverter takes off. And what do you mean by wasted? Where should the energy go? It can hardly dissolve into thin air.

reinhard-brandstaedter, 2023-07-10:
@z-master42
It seems SF does this

```
Panels ==> (SF input limit) SF --(limit output to) |==> battery
                               \-(limit output to) |==> (my controllable limit) hoymiles inverter
```

I can (and am doing so based on my household current usage) control the hoymiles' input limit. But SF is still pushing 600W to the inverter even if it is limited to 234W, and it is definitely draining the battery with 600W if there is no light and the hoymiles is set to 0W limit.
Good question where the energy goes. I've asked myself the same thing what happens to solarpanels that are producing 400W when the inverter is limited to 0W...

nfocke, 2023-07-11:
Makes sense. The SolarFlow and the inverter constantly "negotiate" the drawn current / power. With the inverter trying to draw the maximum possible, the SolarFlow convincing the MPTT that the indented power is the best option... ;)

So, yes, controlling either end of this process would give control over the power drawn. However, we do not know the logic that Zendure uses and how it will try to bring the MPTT into taking a higher power, since it cannot know why the MPTT tracker takes less power than it actually could/should. Hence, controlling/ limiting from the Zendure side seems more intuitive.

z-master42, 2023-07-11:
Of course, it would be nicer if this could be set live directly on SF, but this is currently the only viable solution. It would also be great if in the future the daily values (kWh) were available as a sensor via MQTT and not just the respective power (W).

PSA: MPPT and the T stands for Tracking ;-)

reinhard-brandstaedter, 2023-07-12:
I tested a bit more with limiting on the inverter side. The behaviour is as expected, the output is what the inverter draws up to the max output configured on the SF side. The numbers in the app are sometimes just off and/or lagging behind.
But agreed to do a full "intelligent" steering one will need access to the output limit on the SF side. Esp. in regards to that statement: "If the SF is in between, the power goes into the battery until it is full. After that nothing more is produced."
My (and others) idea is that if the battery is full, and solarpower is generated it should go to the inverter (raise output limit to what is coming in to avoid discharging of battery) and allow a user to dynamically adjust the output.
That way one could easily decide if/how much should be drained from the battery while there is still some solarpower available, given the user also gets all the sensor information needed for that decision.

tschweitzer, 2023-07-28:
This is also what I would be very much interested in, so we can control the energy efficiently.

akaoma, 2024-02-14:
Any news on this topic: output regulation control ???

z-master42, 2024-02-14:
Not official, but you can take a look here: https://github.com/reinhard-brandstaedter

ckrugde, 2024-04-21:
Here is a link to a iobroker project: https://www.npmjs.com/package/iobroker.zendure-solarflow
According to the description, it seams to be possible to control the output limit. I am new to Home Assistant, so I need to deep dive first to get it to work in Home Assistant as well. May be somebody can help.

z-master42, 2024-04-21:
That's "reverse engineered" too.