Zendure/developer-device-data-report #21: Got App Key and Secret, can connect to MQTT broker, no data

reinhard-brandstaedter, 2023-07-14:
Hi,
I successfully created my appkey and secret (registered as developer). I can now connect to mqtt.zen-iot.com with the MQTT Explorer, but I do not see any data/topics coming from my SolarFlow device?
I noticed that SolarFlow is connecting to mq.zen-iot.com (another mqtt broker?) and not the mqtt.zen-iot.com.
trying to connect to mq.zen-iot.com with the dev credentials doesn't work (expected)
Is this supposed to work for SolarFlow as well?

z-master42, 2023-07-15:
You need to subscribe to a topic. On the configuration page of the connection, you must click on advanced and enter _yourusername_/# as the topic and add it.

reinhard-brandstaedter, 2023-07-16:
Got that, figured! Thanks!

closed 2023-07-16 by reinhard-brandstaedter