Zendure/developer-device-data-report #22: mqtt.zen... no values anymore

agrdbf, 2023-07-17:
Since 10 a.m. CEST I don't get any values from mqtt.zen-iot.com anymore. Everything was running the last week without problems.
[image]

Has somebody else the same issue? Is something known about that?

Stephan23579, 2023-07-17:
I have the same issue. Connection OK, but Receiving no messages.

agrdbf, 2023-07-17:
> I have the same issue. Connection OK, but Receiving no messages.

good to know. Thanks for your reply.

Zendure, 2023-07-17:
I apologize for the inconvenience you're experiencing. After investigating
the issue, we found that there has been a surge in device messages leading
to message accumulation on the server. However, please rest assured that we
have taken measures to scale up and resolve this problem.

We sincerely apologize for any inconvenience caused and appreciate your
patience during this time. If you have any other questions or need further
assistance, please don't hesitate to contact us. We are committed to
providing you with excellent service.

Once again, we apologize for any inconvenience caused.

Wishing you a pleasant user experience!

Best regards Zendure Developer Team

agrdbf, 2023-07-17:
> I apologize for the inconvenience you're experiencing. After investigating the issue, we found that there has been a surge in device messages leading to message accumulation on the server. However, [...]

Hi, looks better now. Values from mqtt are coming in again. Thanks for solving the problem that quick!

agrdbf, 2023-07-17:
Issue is solved, can be closed.

jsuelwald, 2023-07-17:
Re-opening the issue, becuase doesn't work here right now; neither with mqtt as bridge nor mit mqtt explorer

z-master42, 2023-07-17:
Can also confirm that it works again. And the update of the values is much faster than before.

z-master42, 2023-07-17:
The AppKey ist your username. The AppSecret ist your Password.

jsuelwald, 2023-07-17:
Ok, sorry - i've got that setup that way - still doesn't work

jsuelwald, 2023-07-17:
[image]

As shown here - worked fine until then..

z-master42, 2023-07-17:
That's crazy. For me, as for everyone else, it stopped working from the same time. But since this afternoon it runs again, as per the Zendure post above.

jsuelwald, 2023-07-17:
Yeah, but here it doesn't and i've got not way to debug that further -.-

jsuelwald, 2023-07-18:
I think i'll wait until the SOlar Flow Module offers it's data locally via MQTT as well, i hate using an external data source.

agrdbf, 2023-07-18:
> I think i'll wait until the SOlar Flow Module offers it's data locally via MQTT as well, i hate using an external data source.

Have you set your subscription to AppKey/# ? (in mqtt explorer in the advanced section)

jsuelwald, 2023-07-18:
Yeah, yesterday it didn't work at all.

Now it works a bit. Gets data and disconnects again
[image]

MQTT-Broker on Home Assistant is configured as follows:

```connection external-bridge
address mqtt.zen-iot.com:1883
remote_username <snip>
remote_password <snipsnipsnip>
clientid zendure

topic <snip>/# both 0 zend/
```
And i didn't touch that config for days, because it was working fine.

agrdbf, 2023-07-18:
I do not use home assistant for that, but from config side it looks good to me. I'm using a similar config with nodered.
But maybe try a more generic clientid. I'm not sure and I'm not a mqtt pro, but I've something in mind that the client-id has to be unique for the whole broker and in the past with the ecoflow mqtt server that was a common issue, when multiple users uses the same clientid.

z-master42, 2023-07-18:
Isn't it then the exact opposite? A more specific clientid? Try remote_clientid <snip> instead.

jsuelwald, 2023-07-19:
Suddenly, as of 21:00 yesterday (CEST) it works again without me changing anything. I can't rule out as issue with internet routing and/or DNS because i didn't think of that as a possible fault at this time.

agrdbf, 2023-08-05:
Looks like the issue occurs again. Since yesterday somewhen in the evening, I don't get values anymore. Looks exactly the same like the last time. Connection to mqtt-zen-iot.com is possible, but no values. Am I the only one or can anybody else confirm?

DeathForce07, 2023-08-05:
> Looks like the issue occurs again. Since yesterday somewhen in the evening, I don't get values anymore. Looks exactly the same like the last time. Connection to mqtt-zen-iot.com is possible, but no values. Am I the only one or can anybody else confirm?

same here

agrdbf, 2023-08-05:
Since 2:30 p.m. it seems to be fixed, since then I get values again

achi87, 2023-08-11:
Nothing again as far as I can see (last 6/7 hours)

agrdbf, 2023-10-03:
Looks like we have the same issue again. Don't get any values from zendures mqtt since 2 hours. :/