Zendure/developer-device-data-report #25: Control of the Output wattage (API/MQTT/Webhooks)

Pixel-Chris, 2023-07-31:
Output wattage control per
webhooks:
Shelly 3EM power meter
…

mqtt:
nodered
…

api:
etc.

steffenxx, 2023-07-31:
Apparently you can set the "outputLimit" via MQTT. In the MQTT Explorer I have seen the following under:

xxxxxxxx/sensor/device/yyyyyyyoutputLimit/config

{
"unique_id": „xxxxxxRroutputLimit",
"state_topic“:“xxxxxxx/yyyyyy/state",
"command_topic": „xxxxxxxx/yyyyyyyoutputLimit/set",
}, "name": "outputLimit",
"value_template":"{{ value_json.outputLimit}}"
}

I just can't get a flow built in NodeRed to set the value with. But am also new to the topic.

RobertLukan, 2023-08-01:
Has anyone tried this option ? To me this is feature is a deal breaker - would I buy this product or not. I am monitoring use of my electricity with Shelly3em(two of them), so I know what is the usage of my electricity. Having an option to control the output of SF would basically compensate the usage(on 3 phases). If we could control this inside of our LAN would be even better, without using cloud MQTT.

steffenxx, 2023-08-01:
Hi Robert, 

I also use a Shelly pro 3EM and want to achieve the same. The API of Zendure seems to be readonly and a direct way to the device is not known at the moment. But I would also be happy if it goes through the cloud MQTT server. Especially since I only use Solar Flow in my guest wifi, because the box probably can not log into a WPA3/WPA2 Fritz.Box wifi because of missing PMF support. I already have a ticket at Zendure. 

It must go via MQTT, when I played around yesterday, I managed with a syntax to set the outputLimit to 0W. Unfortunately I don't know enough about JSON and MQTT to develop this without help.

Pixel-Chris, 2023-08-21:
Does someone have one of these satellite plugs??
Can someone check the communication of these plugs?? The plugs can control the output! So it must be possible…

Pixel-Chris, 2023-08-30:
Something is happening…

[image]

last App updated …

„CT“ and Shelly 3em Pro as PowerMeter

ortegafernando, 2023-10-24:
Any news? thanks

apunkt, 2024-09-21:
As there is no update for almost a year now from @Zendure .
Anyone with workarounds to make it work?