Zendure/developer-device-data-report #28: zendure has stopped development mqtt ?

gh-fx2, 2023-08-20:
Today I tried check mqtt-connectivity. Summary : BAD - it looks like zendure will not develop the mqtt interface anymore.
I notice that the last changes here in the repository are 4 months ago. The MQTT broker software does not seem to have evolved much either. The current interface does not even provide all values. (For example, I can't find anything about my satellite batteries). Set topics is not yet supported either. Maybe Zendure has only one developer for this and he's on vacation? The oldest issue is an good sample of what I mean. ... no useful help for such simple requirement.

For me - I can not recommend buying Zendure components. A (**full funtional**) mqtt interface is essential for use in an automated environment.

Sorry, that's simply bad.

For others - It looks like that it makes no sense to write new topics here. Zendure can not help - or will not help. Both ways are not a good sign.
PS: it is only an subjective impression - like a zero-star customer recession.

steffenxx, 2023-08-20:
Are you sure? I´m still recieving values via MQTT.

jaegerschnitzel, 2023-08-21:
> Are you sure? I´m still recieving values via MQTT.

Yes it's working. I think @gh-fx2 means that the implementation is on a very basic level and the development made no progress in the last 3 months.
I can agree to that. Zendure announced to support local MQTT communication without the use of cloud and also to allow device control (change settings). But there is no progress at all.

steffenxx, 2023-08-23:
ok, you're right. Nothing is happening at Zendure. In my case, I can't get the PV-HUB to connect to my regular WLAN that accepts WPA3 and WPA2. I only get a connection through the guest wifi. I have opened a ticket with Zendure and no matter how I describe it, their response is always the same. "We do not support WPA3". But that's not the problem at all, 80% of my devices don't support WPA3 and log in via WPA2. I suspect that the PV-HUB has a problem with PMF. On the other hand, I may not want to have a device on my WLAN that Zendure can access.

jaegerschnitzel, 2023-08-27:
Just saw the Zendure App update with a few interesting new features:

`1. Added Amup mode to SBV.
2. Add push function.
3. The device supports offline data synchronization.
4. Add a mall module.
5. Added SolarFlow bypass mode.
6. The "Family First" option has been added to the smart matching mode.
7. Added smart CT mode.
8. Satellite Monitor CT access.
9. Shelly 3EM Pro access.

Note: Some new features require SolarFlow 3.0.0 firmware support, which will be released a week later.`

steffenxx, 2023-08-28:
Good news :-) Apparently the developer is back from vacation ;-)

jvinkovic, 2024-02-11:
@Zendure ?

jvinkovic, 2024-03-19:
@Zendure any comment?

mi-hol, 2024-10-29:
> Zendure announced to support local MQTT communication without the use of cloud and also to allow device control (change settings). But there is no progress at all.

18 months passed since Zendure marketing made announcement but development did not deliver any of the promised future features. Can customers develop trust in Zendure?

jvinkovic, 2024-10-30:
Yes. @Zendure is not to be trusted and they proved it many times in last year. That is why we have ecloflow any many others far more reliable and far more understandable.

jvinkovic, 2025-03-06:
Just use ecoflow or anker.