Zendure/developer-device-data-report #29: Call the REST API from Windows

Dr-Sommer, 2023-08-30:
Hi, I'm trying to get my Dada with CURL (downloaded the newest Version" but when I try to get data with: 

`curl -H "Content-Type: application/json" -X POST -d '{"snNumber":"EURE-SERIENNUMMER","account": "EUER-APP-NUTZERNAME"}' https://app.zendure.tech/v2/developer/api/apply`

I get this error:
 
`curl -H "Content-Type: application/json" -X POST -d '{"snNumber":"Serial-Number","account": "APP-Username"}' https://app.zendure.tech/v2/developer/api/apply`

Could somebody help me with this? ThX in advance

z-master42, 2023-09-04:
Unter Windows sieht der Befehl so aus:

`curl -i -v --json "{'snNumber': 'EureHubSeriennummer', 'account': 'EureEmailadresse'}" https://app.zendure.tech/v2/developer/api/apply`

Jumhansen, 2023-10-16:
@z-master42 
Erstmal Danke für deine Anleitung um den Solarflow per mqtt aulesen zu können. Ich will das ganze allerdings im iobroker machen. Jetzt zu meinem Problem.
Ich schaffe es einfach nicht mit dem curl Befehl (Seriennummer und Emailadresse sind eingetragen) einen appkey und appsecret zu bekommen. Nach absetzen des curl Befehls erhalte ich folgende Meldung:

* Trying [2600:9000:2250:4800:d:d112:8200:93a1]:443...
* Connected to app.zendure.tech (2600:9000:2250:4800:d:d112:8200:93a1) port 443 (#0)
* schannel: disabled automatic use of client certificate
* ALPN: offers http/1.1
* schannel: next InitializeSecurityContext failed: Unknown error (0x80092012) - Die Sperrfunktion konnte keine Sperrprüfung für das Zertifikat durchführen.
* Closing connection 0
* schannel: shutting down SSL/TLS connection with app.zendure.tech port 443
curl: (35) schannel: next InitializeSecurityContext failed: Unknown error (0x80092012) - Die Sperrfunktion konnte keine Sperrprüfung für das Zertifikat durchführen.

Hättest du da eine Idee für mich. Voarb schon mal vielen Dank.
Jumhansen

Jumhansen, 2023-10-16:
@z-master42 
vielen lieben Dank für deine Antwort. Ich konnte mittlerweilen meinen appkey und appsecret bekommen. War mir
über den Windows PC einfach nicht gelungen, bin leider auch kein Informatiker. ;-)
Über das online Tool REQBIN konnte ich den curl Befehl dann erfolgreich absetzen, mit Antwort appkey und appsecret.
Werde das am späten Nachmittag zuhause im mqtt Adapter im iobroker ausprobieren.
Nochmals vielen lieben Dank.
LG Jumhansen