Zendure/developer-device-data-report #30: Change the Endpoint in the APP, but how in MQTT??

tasisale, 2023-10-06:
i have change the Endpoint from "Global" to Germany, now my MQTT Endpoint shows:
`{"code":400,"success":false,"data":{},"msg":"Please bind the device first"}f`
whats the Endpoint: https://app.zendure.tech/v2/developer/api/apply <- for Germany???

thx and regards
simon

FloSchl8, 2023-10-07:
Same problem here (also from Germany)
Been trying since yesterday and double checked account and serial
Still getting `Please bind the device first`

z-master42, 2023-10-07:
Same here. @Zendure ?

ruvnax, 2023-10-07:
Same here. @Zendure ?
Changed Endpoint to GER , no MQTT

Please fix

Zendure, 2023-10-09:
Sorry to keep everyone waiting for so long.
Today we launched the EndPoint for the open platform in the European region. Everyone can use the corresponding EndPoint to apply for permissions.

Zendure, 2023-10-09:
You can reauthorize by calling the corresponding API for the following situations:

1.If you forget the appKey and secret for the MQTT connection.
2.If you switch from the global server to the European region.

tasisale, 2023-10-09:
> You can reauthorize by calling the corresponding API for the following situations:
> 
> 1.If you forget the appKey and secret for the MQTT connection. 2.If you switch from the global server to the European region.

how i can do this, please post a hotwo, step for step.

thx
simon

tasisale, 2023-10-09:
ok, have fount it in yout home-description.
it is -> Endpoint(Europe): https://app.zendure.tech/eu/developer/api/apply

Zendure, 2023-10-09:
Could you please confirm if your issue has been resolved? To create or retrieve a previously authorized account, please use your app account and device SN to call the European site API (https://app.zendure.tech/eu/developer/api/apply).

ruvnax, 2023-10-09:
> Could you please confirm if your issue has been resolved? To create or retrieve a previously authorized account, please use your app account and device SN to call the European site API (https://app.zendure.tech/eu/developer/api/apply).

For me it works , i recive data from MQTT but with a little latence ... thx

steffenxx, 2023-10-12:
When I send the API request to https://app.zendure.tech/eu/developer/api/apply I get the following response:

{
    "code": 400,
    "success": false,
    "data": {},
    "msg": "Please bind the device first"
}

Do I need to rebind my Solar Flow?

Zendure, 2023-10-12:
Dear Developer,

Thank you for reaching out to us regarding the API request for your Solar
Flow device. We understand that you received the following response:
{
"code": 400,
"success": false,
"data": {},
"msg": "Please bind the device first"
}
Based on this response, it appears that you may need to rebind your Solar
Flow device.

If you were previously using the global service site, and now you received
the API response from the European server (
https://app.zendure.tech/eu/developer/api/apply), it is necessary to bind
your device to the European server. This means that you will need to follow
the instructions provided for binding or reconfiguring your device to the
European server. Once the binding process is complete, you can proceed to
call the API interface again to obtain the desired results.

If you require any further assistance or have any additional questions,
please feel free to reach out to us. We are here to help.

Thank you for your cooperation.

Best regards,

Zendure Developer Team

steffenxx, 2023-10-12:
I removed and reconnected my Solar Flow via the Zendure app. I could not select a European server at any point. As a result, my Solar Flow is still connected to the Global Server and I get the "Please bind the device first" response from the EU API.

How must the rebind exactly be done?

z-master42, 2023-10-12:
Go to the account settings and change the region to your European country. Unbind the SolarFlow beforehand.

steffenxx, 2023-10-12:
thx, it works :-)