Zendure/developer-device-data-report #31: Solarflow current Battery temperature is missing via MQTT

AzraelsDisk, 2023-11-13:
In the Zendure app, the temperature of each battery is shown, but this information is missing on MQTT. Only a max temp is provided, but this is not the current one. 
[image]

AzraelsDisk, 2023-11-15:
> True, unfortunately, Zendure has apparently abandoned any further development. Very disappointing.

Oh, I haven’t seen this…. That’s really disappointing…

Zendure, 2023-11-15:
The battery pack information has been updated. Developers can perform data parsing on the packData object as needed. Field explanation:

socLevel: Battery level
maxTemp: Battery temperature
maxVol: Maximum battery voltage
minVol: Minimum battery voltage
sn: Battery serial number

AzraelsDisk, 2023-11-15:
Yes but is max temp the current one? Or the max of the day?

Zendure, 2023-11-16:
maxTemp refers to the current highest temperature of the battery cells inside the battery pack. It represents the present temperature rather than the maximum temperature recorded throughout the day.

AzraelsDisk, 2023-11-16:
Thanks for the explanation, but why as an example is on the mqtt 29 degrees reported and in the app 5,9 degrees?

Zendure, 2023-11-17:
It seems like there might be a discrepancy between the temperature values reported on MQTT and the app. The maxTemp reported on MQTT is in Kelvin, so you would need to convert it to Celsius or Fahrenheit according to your preference.

ragman11, 2023-12-02:
Is it possible to create a sensor for the maxTemp, to extract it from the packData?

z-master42, 2023-12-02:
Yes, that's possible.

ragman11, 2023-12-02:
and how the entry look like in the mqtt.yaml. I tried, but could not find a solution.

z-master42, 2023-12-02:
I have a concrete idea, but have not yet been able to implement and test it due to a lack of time. Hopefully within the next few days.

ragman11, 2023-12-04:
thank you for the code, works perfect

z-master42, 2023-12-05:
Is explained under Steps to Subscribe to Device Data.

z-master42, 2023-12-17:
Maybe you have to force it. When I click on the battery icon in the SolarFlow overview in the Zendure app so that it shows the charge level and temperature of the individual batteries, all MQTT values are updated, even those that have not yet been transmitted.

z-master42, 2023-12-17:
As I wrote in my previous post, my MQTT values (including everything that comes via packData) are initially updated when I go to the battery detail view in the app.

leow149, 2026-01-09:
Hi, is there any news on this since it is not closed? Seems like at least the SF 800 Pro still only reports the "hyper" temperature instead of the individual battery pack temperatures