Zendure/developer-device-data-report #34: possibility to regulate power

jvinkovic, 2024-01-20:
Can you please add possibility to regulate power? 
Or possibility to set up local mqtt and regulate power?

gitmachtl, 2024-02-11:
If i can jump on that Issue, would be very nice and helpful to make the `acInputLimit` parameter for the Superbase V setable via the API or even better via a local REST-API directly on the device.

akaoma, 2024-02-14:
Same as @gitmachtl, a regulation parameter for SolarFlow / Superbase V would be really nice! :)

jvinkovic, 2024-03-02:
@Zendure so?

ps915, 2024-03-17:
That's the only thing which is stopping me from buying a lot of Zendure stuff ;)

jvinkovic, 2024-03-17:
You should never buy zendure. Do not deliver on any promises.
Too much problems (there have not been a week without some problems), and non existent support. Ecoflow is much better and much better customer support.

ps915, 2024-03-17:
Thanks! So sad! I will take a look at ecoflow storage systems then! 

Bye Bye Zendure! =)