Zendure/developer-device-data-report #35: MQTT not sending 0 values

thierrywehr, 2024-01-25:
Hello
i'am using home assistant mqtt and only non zero data from solarflow are received
to be more precise an example
get data for entitie packInputPower when it is above 0
as soon as it drops to 0 ( no power ) mqtt is not updating Home Assistant
any idea to correct this

Thierry

Zendure, 2024-02-29:
When it drops to 0, is the device still online? If the device remains online, not in standby mode, and there are data changes, MQTT should be able to update the data in real-time.

thierrywehr, 2024-02-29:
yes the device is still online
i have discovered that the only way to make it send updates is to go in the
zendure mobile application and confirm existing "rendez vous" configuration
which will enforce a flush of all full mqtt datas including voltage for
example
the strangness is that when the zendure application shows the solarflow
offline i can still see mqtt updates coiming from the zendure cloud and
updating but all solarflow configuration is stucked
this means that the solarflow HUB is still connected to internet but the
application is not able to reach it
i can clearly see issues with the zendure cloud which impact the usage of
the solraflow
the best would be to be able to send direct mqtt actions to the solaflow
and bypass the zendure cloud which is buggy and overbooked
i can demonstrate that since last few days, datas in zendure cloud are
wrong despite the fact that it is working smoothly

stefan-peters, 2024-07-01:
I can confirm the same problem. The last "0" is not always send and the values remain on the last non-0 value.

leop800, 2024-08-04:
I have the same problem, no zero value and they remains with last value

drschnalli, 2024-09-02:
I also got the same issue. Output pack power still shows 36 Watts even its not outputting anything :(

fr43nk, 2025-02-09:
Hi,
I also have such an issue. Values like
- packInputPower
- outputPackPower
- outputHomePower

Do not update correctly. outputHomePower and packInputPower keep their value even it should be zero. OutputPackPower does not send a zero update value.