Zendure/developer-device-data-report #37: MQTT interface needs to deliver more complete information for Solarflow and attached batteries

ppitkin, 2024-03-06:
The current MQTT interface needs to be expanded to include additional information such as:

- The type of battery (e.g. AB1000 or AB2000)
- The current and maximum capacity of the battery (not just the State of Charge)
- If the hub has gone offline
- The energy plan that has been selected on the app.
- The Energy generated (Wh), The Energy consumption (Wh), The energy to and from the battery(s)

The interface should send at regular intervals (say every 15 mins)  a single update with all current values - rather than just changed values.

goscho-biz, 2024-03-06:
especially those

> The Energy generated (Wh), The Energy consumption (Wh), The energy to and from the battery(s)

Zendure, 2024-03-08:
The battery type can be determined by the battery code attribute, where those starting with A are AB1000, and those starting with C are AB2000. We will consider gradually supplementing other issues. 👍