Zendure/developer-device-data-report #38: MQTT interface sending undocumented values

ppitkin, 2024-03-07:
I just by chance that the interface sent yesterday a value for an undocumented parameter when the solar power went down to zero. I think the field was call something like hubStatus and it had a value of 1.

Please can you ensure that the documentation (and the definition of the parameneters under the  /device and /status topics accurately reflct the information that is or can be sent when subscribed to a Solarflow device.

Zendure, 2024-03-08:
This is a newly added parameter in the latest firmware. The documentation has now been updated to include this parameter. Thank you for your attention to detail!

ppitkin, 2024-04-18:
Since the recent firware update new values are beinig delivered in the MQTT interface. Please update the documentation so that we can understand what they represent.

I have seen at least the follwing two new attributes being sent:

"wifiState":true
"inverseMaxPower":1200

I would also point out that you are still not sending the totalVol value in the feed. Please add this ASAP.