Zendure/developer-device-data-report #39: Add the new AIO device to the MQTT interface

ppitkin, 2024-03-07:
As with my Solarflow devices, I need the ability to monitor and intergrate my AIO device into my home automation systems. As the AIO is now available on the market the MQTT interface should deliver all the available paramerer information and values for AIO devices attached to my account.

It is important that the list of parameters published should reflect the full scope of the information that is available via the APP.

Of particular interest would be, not only the information requested to be added to the Solarflow parameter list, but also things like the output adjustment values that the device is making when in the energy mode of adjusting the output based on the information its receiving from the Shelly 3EM (or other) grid meter.

This would help greatly when develoing automation scrits to make optimal use of available Solar and Battery power.

Zendure, 2024-03-08:
The outputLimit attribute will report the value of real-time adjustments to the electricity usage plan, and the outputHomePower is the actual power output to the home.

razorlikes, 2024-03-20:
I don't seem to get any data from my AIO 2400 either. Any updates on when this will be implemented? @Zendure

martinh78, 2024-03-25:
@Zendure I also need the AIO Data via MQTT. When will this be available?

Fry3199, 2024-04-01:
@Zendure Please provide the MQTT data via the Zendure MQTT broker or - even better - that the AIO can use a local MQTT broker.

FuwoWolf, 2024-04-05:
I'd also like to throw my Hat on the pile and ask @Zendure to delivere the AIO data through the MQTT and make it local (Smart Home integration was one of the reasons I chose Zendure).
According to my Ubiquiti UDM the AIO is already running a MQTT Stream (although sometimes Unifi Traffic Identification is off)

MimbaMonkeyHouse, 2024-06-06:
+1 as I need this too

Ken-dono, 2024-07-06:
+1 as I also would love to see this !

towe75, 2024-07-11:
They added ACE and Hyper specific fields to the Readme a few days ago:  https://github.com/Zendure/developer-device-data-report/commit/f24cf68032bf9cdc9d367bee07c991d5bf7e1130

Maybe it will be supported soon. Does it already work for somebody here?

razorlikes, 2024-07-11:
> Does it already work for somebody here?

Just tried, I don't get anything.

noide35, 2024-09-17:
It seems it was added no ? I just tried today to use MQTT Explorer with the info related to my AiO2400 and I get data !

A quick extract of one message in json format :

{"outputHomePower":740,"packInputPower":776,"electricLevel":40,"packData":[{"minVol":323,"sn":"<hidden>"}],"sn":"<hidden>"}

tixoman, 2024-09-18:
Yesssss I also get data from mqtt**-eu**.zen-iot.com finally

noide35, 2024-09-20:
So new that some data are there, would be great to get the next step :
- Get a currently missing data : The energy plan information (which one is selected)
- Device Control through MQTT (including output-limit, energy plan selection, ...)