Zendure/developer-device-data-report #40: URGENT! SHELLY integration broken. Not SECURE! Too much power provided

jvinkovic, 2024-03-13:
Solarflow IS BROKEN... 

Shelly works fine and shows correct realtime consumption. 

ZENDURE IS CONSTANTLY PROVIDING TOO MUCH POWER!

Zendure master solarflow firmware is v2.0.38
Battery AB1000 BMS is v2.0.22

From what I saw, it seems that it sums absolute value...
E.g.
phase 1 is -213 W
phase 2 is 122 W
phase 3 is 323W
It will sum it up and show that consumption is 658W instead 232W (it sums up only absolute values) and then it will provide more power and then it will have greater sum and so on until provides maximum it can.

[image]

ppitkin, 2024-03-19:
I met with the zendure team a couple of weeks ago and we spent several hours discussing the topic of 3 Phase systems as used in many European countries. They are still on the learning curve here and this is why the behaviour is not what we might be expecting.

The other thing to remember is that they are using the power meter (Shelly 3EM in this case) to control the hub and not the inverter. This means that they will change the hub output and then the MPPT's on the inverter will then make another adjustment to optimise the power and then feed something different into the power network. 

On my system I user the meter to adjust the inverter (Hoymiles) and this seems to work fairly well.

jvinkovic, 2024-03-19:
1, 3, or 500 phases it does not matter. If shelly is saying consumption is X then it does not matter which phase is it or how many so this is just excuse for botched job and/or they do not know how to sum values and does not want to communicate without cloud.  

PROBLEM is that shelly is reporting all fine, but zendure is making completely wrong calculations and control. 
Solution is communication directly through local network which is supported out of the box in shelly. Zendure is the one making it complicated. 
If I known this would go with ecoflow. WAAAAY better platform, waaay better customer support, nothing of such excuses and such problems with simple solution.

ppitkin, 2024-03-19:
Hi thiettywehr,
in response to your concern regarding what will happen if you have the Hub set to output 800W and the inverter says it only needs 400W - the answer is nothing.

The inverter will only allow it MPPT inputs to draw 400W from the HUB. The HUB will then only deliver 400W as that is the maximum draw on the output of the hub. If the hub is receiving more than 400W from the solar pannels it will try to push the extra into the Batteries. Should the batteries be full then it will use its MPPT's to throttle the power drawn from the solar pannels.

I have spent months testing and trying to find the best setup for my particular needs. I actually have 3 Solarflow hubs which feed into a single Hoymiles 3 phase microinverter (MHT-2250). The inverter is set up to adjust its output to ensure that there is 0 Watts fed into the grid. This is done by using a meetering device and the Hoymiles DTU  (the opensource hub can also be used). This setup works great and without any additional coding in the background or the need to use any of the Zendure HUB functionality. You simply configure the solarflow hub to output the full 1200W on its output.

I did a lot of testing on the AIO unit before it was publicly released and there I had the same issues as jvinkovic is indicating. The throttling of the output to give zero grid input is not very good. I tried all sorts of combinations trying both to balance a single phase as well as all three phases. Nothing worked very well.

I'm sure the Zendure team will get it right in the end. They just seem to be a bit slow to react and fix things and as I mentioned above, a 3 phase electrical setup is not well understood by them.

thierrywehr, 2024-03-19:
Hello Peter

that sounds good
at the time i am using mqtt setting output power through it
works fine with home assistant and shelly equipment
i hope zendure will very soon make available mqtt local connectivity to
avoid using specific setups :)

thank you
Thierry

ppitkin, 2024-03-20:
there is the open source version of the DTU (OpenDTU and AhoyDTU). Here you can either build it yourself (around 10€) or by a finished product (around 50€). I went for the latter as I wanted to save some time. I have to admit though that I do not use that device to control the inverter. I user the Hoymiles DTU-PRO-S and the Holymiles web app to set it up for zero export to the grid. I think you can do the same with the open source DTU - I have just never tried it. The big advantage of the open source DTU is that it supports MQTT

Be carefull to get the correct DTU for the version of the Hoymiles inverter that you are using. They work on different frequency bands. The version of the OpenDTU that I purchased actuall supports both network bands and is linked to all my hoymiles inverters (4 in total , various different models).

thierrywehr, 2024-03-20:
Hello Peter

i had a close look at the way it works
it is not the HUB which does the regulation on passthrought mode it is the
inverter
as panel circuit is directly connect to the hoymiles, it is the mppt
hoymiles which is managing the power
which means that when using for example APSystems one you will get all
power from the panels
let's say you have a HUB2000 with 1600 Wc solar panel and a APSystems DS3
which can handle 880 Wc, when in pass mode you will get all the 1600 wc
from the solar panel to the Micro Inverter
i think that Zendure have to clarify or soon they will have issues with
customers and espcially with the HUB2000