Zendure/developer-device-data-report #41: Homeassistant Batterysystem

KevinKrueger, 2024-03-16:
I have now spent hours integrating the battery input & output into the dashboard and it has not worked. There are thousands of instructions that don't work. Yes.... is just annoying.
Does anyone happen to have the code for the yaml that converts the mqtt sensor into an energy sensor so I can add it to the energy dashboard?

Thanks in advance. 
Kevin

thierrywehr, 2024-03-16:
You get only power no energy from mqtt
So you need to créate an entity that count power to make energy data
Google ans check for "home assistant energy entity from power entity"

ppitkin, 2024-03-19:
I did submit a request to have them add this information to the feed. They have it in their App and so there should be no reason why they cannot deliver it on the MQTT feed.

It unfortunately takes the Zendure team forever to implement any sort of improvement

z-master42, 2024-04-17:
Use the Riemann Summen Integral in the helper area of the Home Assistant. Method is "left". Prefix is "kilo". Done.