Zendure/developer-device-data-report #44: MQTT ist broken, some data missing

Hans1Hansen, 2024-04-09:
Two days ago I could not receive any data for MQTT, like a lot of other users.
Today some data came in again, but three important values for the solar panels are still missing:
SolarPower1
SolarPower2
SolarInputPower
In MQTT Explorer all topics have the same number of messages, even the elements for SolarPower. 
But in the element "state", where the values are transmitted (in my opinion), all values (e.g. electricLevel, outputState, outputHomePower) are processed exept the three missing ones.
Could you please fix this issue? Thanks.

Cheers, Hans

turricanus, 2024-04-11:
Here is it a little different, I only don't get "SolarInputPower",I've made a workaroud and sum Solarpower1 and Solarpower2.
Please fix it.

Hans1Hansen, 2024-04-11:
> Here is it a little different, I only don't get "SolarInputPower",I've made a workaroud and sum Solarpower1 and Solarpower2. Please fix it.

Since yesterday about 12 o'clock things changed to mee, too. Two values now come in, SolarInputPower is still missing.
In Home Assistant I did not sum the two values but made two different entries, so the graph now shows my two pabels in one column with different colors.

turricanus, 2024-04-11:
> Since yesterday about 12 o'clock things changed to mee, too. Two values now come in, SolarInputPower is still missing. In Home Assistant I did not sum the two values but made two different entries, [...]

This works for me:
unit_of_measurement: "W"
    device_class: "energy"
    value_template: "{{ value_json.solarPower1 | int(0) + value_json.solarPower2 | int(0) }}"

I've duplicated the original sensor with a new name and uid to look if the data comes back.

Zendure, 2024-04-11:
@turricanus You are Right，solarInputPower is the sum of solarPower1 and solarPower2. Reporting of this value has now commenced.

turricanus, 2024-04-11:
> @turricanus You are Right，solarInputPower is the sum of solarPower1 and solarPower2. Reporting of this value has now commenced.

Now, the Value of solarPowerInput are back. Very good thanks

Hans1Hansen, 2024-04-11:
Very nice, I can confirm this, too. 
I wonder why it did ever stop.

phenomeus, 2024-06-17:
@nerra0pos maybe this? https://github.com/reinhard-brandstaedter/solarflow-control

dav1dBoy, 2025-03-07:
Hi，Would you consider the new architecture for Zendure products? It is currently in internal testing. (https://github.com/dav1dBoy/zenSDK)

jgrieger, 2025-03-08:
> Hi，Would you consider the new architecture for Zendure products? It is currently in internal testing. (https://github.com/dav1dBoy/zenSDK)

This would not help for pre-2025 Zendure products, would it? I am not willing to buy a new Zendure device to finally have an API available via LAN (without hijacking the mqtt connection).

dav1dBoy, 2025-03-12:
> > 您好，您会考虑 Zendure 产品的新架构吗？目前处于内测阶段。( https://github.com/dav1dBoy/zenSDK )
> 
> 这对 2025 年之前的 Zendure 产品没有帮助，不是吗？我不愿意购买新的 Zendure 设备，以便最终通过 LAN 获得 API（而无需劫持 mqtt 连接）。

> > Hi，Would you consider the new architecture for Zendure products? It is currently in internal testing. (https://github.com/dav1dBoy/zenSDK)
> 
> Great thing, but what about the other Zendure products except SolarFlow 800?

> This would not help for pre-2025 Zendure products, would it? I am not willing to buy a new Zendure device to finally have an API available via LAN (without hijacking the mqtt connection).

 I hope you will like the new product, SolarFlow 800 (and consider purchasing it). I put a lot of effort into designing ZenSDK.

dav1dBoy, 2025-03-12:
> This would not help for pre-2025 Zendure products, would it? I am not willing to buy a new Zendure device to finally have an API available via LAN (without hijacking the mqtt connection).

I hope you will like the new product, SolarFlow 800 (and consider purchasing it). I put a lot of effort into designing ZenSDK.

AMTQ, 2025-03-12:
> Just use ecoflow or anker. Zendure is shit

Do they have local API?

dav1dBoy, 2025-03-13:
> > Just use ecoflow or anker. Zendure is shit
> 
> Do they have local API?

YES ！ Would you consider the new architecture for Zendure products? It is currently in internal testing. (https://github.com/dav1dBoy/zenSDK)

thierrywehr, 2025-03-13:
Good afternoon

this makes non sens because of mqtt in HUB1200 - HUB2000 - HYPER2000
just give all informations about mqtt and a way to enforce mqtt address server from application

zendure as to apply to his own assets which were about home assistant local connexions

can you explain all logs from mqtt

/73bkTV/<own_id>/log{"deviceId":"<own_id>","logType":2,"timestamp":1722891671,"log":{"sn":"SN_NUMBER","params":[79,2,795,792,0,0,4940,4940,0,0,-19,-17,0,0,329,329,0,0,329,329,0,0,0,0,0,0,0,0,0,0,8449,257,0,0,26,25,0,0,882,1094,0,0,490,0,0,170,441,178,0,17000,7,268447777,0,0,0,0,0,0,0,0,0,1,0,0,0,0,0,1,0,106,163,1,1,0,1,1,0,0,0,0,0,1,0,0,0,10,0,512,0,0,0,0,0,0,0,0,0]}}

and all commands like

iot/73bkTV/<own_id>/function/invoke{"arguments":[{"autoModelProgram":2,"autoModelValue":1245,"msgType":1,"autoModel":9}],"deviceKey":"<own_id>","function":"deviceAutomation","messageId":87396534,"timestamp":1728220004}