Zendure/developer-device-data-report #46: MQTT strange behaviour on value for solarInputPower since firmware update

ppitkin, 2024-04-18:
I am observing strange behaviour on the "solarInputPower" values being delivered by the MQTT interface.

Like many users I had a number of days where no values were being sent for solarPower1, solarPower2 and solarInputPower.

This seems to have been corrected, but now I see problems with the solarPowerInput field.

Previously the solarPowerInput field was always delivered in a message that also contained the solarPower1 and solarPower2 fields. 
The value of the solarPowerInput field was always the sum of the solarPower1 and the solarPower2 fields (as would be expected)
That behaviour seems to have now changed. I agree it is not madatory for all 3 values to appear in the same message - but the values don't appear to be consistant.

In the last couple of days the solarPowerInput field is being delivered in a message almost every second. That message normally only contains the SN as a secondary field.

In addition, messages arrive with the solarPower1, solarPower2 and solarInputPower fields (as before) with the value of  solarPowerInput = solarPower1 + solarPower2

Furthermore, messages are arriving with just either solarPower1 OR solarPower2, together with solarInputPower - where the value of solarInputPower is set equal to just the supplied value of solarPower1 (or 2).

For messages where solarInputPower is being delivered WITHOUT a solarPower1 or solarPower2 value, then the value supplied is very different from that of the other messages (difference of several 100 watts in a few millisoconds)

Please investigate.

thanks
Peter.

Here are some example to ilustarate what is happening:

14:02:58.389 | info |   {"solarInputPower":536,"sn":"xxxxxxx"}
14:02:59.344 | info |   {"solarInputPower":534,"sn":"xxxxxxx"}
14:03:00.273 | info |   {"solarInputPower":531,"sn":"xxxxxxx"}
14:03:01.401 | info |   {"solarInputPower":529,"sn":"xxxxxxx"}
14:03:02.648 | info |   {"solarInputPower":526,"sn":"xxxxxxx"}
14:03:04.154 | info |   {"solarInputPower":524,"sn":"xxxxxxx"}
14:03:04.983 | info |   {"solarInputPower":521,"sn":"xxxxxxx"}
14:03:06.090 | info |   {"solarInputPower":519,"sn":"xxxxxxx"}

14:03:09.130 | info	 {"solarPower2":252,"electricLevel":16,"solarInputPower":252,"packData":[{"minVol":329,"sn":"xxxxxxxxx"}],"sn":"xxxxxxxx","outputPackPower":466}
14:03:09.960 | info |  {"solarPower1":239,"solarInputPower":239,"sn":"xxxxxxxx"}
14:03:10.685 | info |  {"solarInputPower":512,"sn":"xxxxxxxxx"}
14:03:11.613 | info |  {"solarInputPower":510,"sn":"xxxxxxxxx"}
14:03:12.518 | info |  {"solarInputPower":509,"sn":"xxxxxxxxx"}
14:03:13.185 | info |  {"solarInputPower":508,"sn":"xxxxxxxxx","remainOutTime":7680}

14:07:15.744 | info |   {"electricLevel":16,"solarInputPower":914,"packData":[{"sn":"xxxxxxxx"},{"maxVol":328,"sn":"xxxxxxxx"}],"sn":"xxxxxxxx"}
14:07:19.234 | info |  {"outputHomePower":783,"solarPower2":430,"packInputPower":52,"solarPower1":417,"solarInputPower":847,"sn":"xxxxxxxx","remainOutTime":59940,"outputPackPower":108}
14:07:19.269 | info |   {"solarInputPower":890,"sn":"xxxxxxxx"}
14:07:21.500 | info |   {"solarInputPower":880,"sn":"xxxxxxxx"}

14:08:16.135 | info |   {"outputHomePower":783,"solarPower2":495,"electricLevel":16,"solarInputPower":495,"packData":[{"minVol":322,"sn":"xxxxxxxx"}],"sn":"xxxxxxxx","outputPackPower":195}
14:08:17.636 | info |    {"solarPower1":495,"solarInputPower":495,"sn":"xxxxxxxx"}
14:08:18.360 | info |    {"solarInputPower":459,"sn":"xxxxxxxx","remainOutTime":97}
14:08:18.519 | info |    {"solarInputPower":457,"sn":"xxxxxxxx"}
14:08:19.362 | info |    {"solarInputPower":455,"sn":"xxxxxxxx"}
14:08:21.200 | info |    {"solarInputPower":454,"sn":"xxxxxxxx"}

Zendure, 2024-04-27:
This issue has been fixed.

ppitkin, 2024-04-29:
Unfortunately the issue appears NOT to be resolved.

please see the logging of the MQTT information being received.

It appears the solarInputPower field has now been removed from the block where the solarPower1 and solarPower2 fields are sent - but the numbers do not match up (i.e. the solarPower1+ solarPower2 <> solarInputPower)

9:02:42.536 | info |  {"solarInputPower":151,"sn":"xxx"}
-- | -- | --
09:02:44.551 | info |  {"solarInputPower":150,"sn":"xxx"}
09:02:48.573 | info |  {"solarInputPower":149,"sn":"xxx"}
09:03:25.550 | info |  {"outputHomePower":20,"solarPower2":82,"electricLevel":50,"solarPower1":70,"packData":[{"sn":"yyy"},{"sn":"yyy"}],"sn":"xxx","outputPackPower":125}
09:03:45.970 | info |  {"solarInputPower":148,"sn":"xxx"}
09:03:54.046 | info |  {"solarInputPower":149,"sn":"xxx"}
09:04:05.314 | info |  {"solarInputPower":147,"sn":"xxx"}
09:04:06.099 | info |  {"solarInputPower":144,"sn":"xxx"}
09:04:07.218 | info |  {"solarInputPower":140,"sn":"xxx"}
09:04:08.131 | info |  {"solarInputPower":136,"sn":"xxx"}
09:04:09.151 | info |  {"solarInputPower":132,"sn":"xxx"}
09:04:10.173 | info |  {"solarInputPower":127,"sn":"xxx"}
09:04:11.191 | info |  {"solarInputPower":122,"sn":"xxx"}
09:04:12.213 | info |  {"solarInputPower":116,"sn":"xxxx"}
09:04:13.263 | info |  {"solarInputPower":113,"sn":"xxx"}
09:04:14.244 | info |  {"solarInputPower":111,"sn":"xxx"}
09:04:17.310 | info |  {"solarInputPower":113,"sn":"xxx"}
09:04:18.471 | info |  {"solarInputPower":115,"sn":"xxx"}
09:04:19.339 | info |  {"solarInputPower":117,"sn":"xxx"}
09:04:20.373 | info |  {"solarInputPower":120,"sn":"xxx"}
09:04:21.386 | info |  {"solarInputPower":121,"sn":"xxx"}
09:04:22.405 | info |  {"solarInputPower":124,"sn":"xxx"}
09:04:23.424 | info |  {"solarInputPower":125,"sn":"xxx"}
09:04:24.431 | info |  {"solarInputPower":127,"sn":"xxx"}
09:05:47.058 | info |  {"solarInputPower":148,"sn":"xxx"}
09:06:02.171 | info | {"solarInputPower":149,"sn":"xxx"}
09:06:03.168 | info |  {"solarInputPower":150,"sn":"xxx"}
09:06:05.215 | info |  {"solarInputPower":151,"sn":"xxx"}
09:06:17.299 | info |  {"packData":[{"socLevel":51,"sn":"yyy"}],"sn":"xxx"}
09:06:26.338 | info |  {"outputHomePower":27,"solarPower2":77,"electricLevel":50,"packData":[{"sn":"yyy"},{"minVol":331,"sn":"yyy"}],"sn":"xxx"}
09:06:27.373 | info |  {"solarPower1":70,"sn":"xxx","outputPackPower":114}
09:06:57.097 | info |  {"solarInputPower":152,"sn":"xxx"}
09:07:13.543 | info |  {"solarInputPower":153,"sn":"xxx"}
09:07:22.669 | info |  {"packData":[{"maxTemp":286.1,"sn":"yyy"}],"sn":"xxx"}
09:07:26.638 | info |  {"outputHomePower":21,"solarPower2":80,"electricLevel":50,"solarPower1":71,"packData":[{"sn":"yyy"},{"sn":"yyy"}],"sn":"xxx","outputPackPower":123}
09:07:39.649 | info |  {"packData":[{"maxTemp":285.1,"sn":"yyy"}],"sn":"xxx"}
09:07:48.753 | info |  {"packData":[{"maxTemp":286.1,"sn":"yyy"}],"sn":"xxx"}
09:08:03.469 | info |  {"solarInputPower":154,"sn":"xxx"}
09:08:15.126 | info |  {"solarInputPower":155,"sn":"xxx"}
09:08:15.879 | info |  {"solarInputPower":156,"sn":"xxx"}
09:08:16.894 | info |  {"solarInputPower":157,"sn":"xxx"}
09:08:18.931 | info |  {"solarInputPower":158,"sn":"xxx"}
09:08:27.004 | info |  {"outputHomePower":15,"solarPower2":81,"electricLevel":50,"packData":[{"sn":"yyy"},{"sn":"yyy"}],"sn":"xxx","outputPackPower":130}
09:08:28.011 | info | {"solarPower1":72,"sn":"xxx"}
09:09:27.190 | info | {"outputHomePower":17,"solarPower2":85,"electricLevel":50,"solarPower1":73,"packData":[{"maxVol":332,"sn":"yyy"}],"sn":"xxx","outputPackPower":132}

So if I just summarize those values being received, and calculate the current value of PV1+PV2 (based on the last values received for these entries I get something like this:

PV1	      PV2	      PV1+PV2	       solarInputPower
			                               151
			                               150
			                               149
70	      82	      152	
		              152	               148
		              152	               149
		              152	               147
		              152	               144
		              152	               140
		              152	               136
		              152	               132
		              152	               127
		              152	               122
		              152	               126
		              152	               113
		              152	               111
		              152	               113
		              152	               115
		              152	               117
		              152	               120
		              152	               121
		              152	               124
		              152	               125
		              152	               127
		              152	               148
		              152	               149
		              152	               150
	        77	      147	               151
70		              147	
		              147	               152
		              147	               153
71	         80	      151	
		              151	               154
		              151	               155
		              151	               156
		              151	               157
		              151	               158
	          81	      152	
71		              152	
73	          85	      158	

As you see there are large discrepancies.

What I expect to see (and was the previous behavior up until you started having the outage and then the firmware update) is as follows:
solarpower1, solarpower2 and solarInputPower should always be updated together and ideally sent in the same MQTT update.
solarInputPower should always be the sum of solarPower1 + solarPower2
You should never get updates for solarInputPower without the associated solarPower1 and solarPower2 values.

Please try to revert the way its calculating and sending data back to as it was previously - as it was working fine at the beginning.

Many thanks