Zendure/developer-device-data-report #47: Improvements in value transmission of the power sensors

z-master42, 2024-04-21:
Since the last major failure of the MQTT brokers, I think it was the weekend before last, the four power sensors for the SolarFlow (Solar Input Power, Output Home Power, Pack Input Power, Output Pack Power) remain at their last value and do not go to 0 W in a defined way. This leads to the batteries being charged and discharged at the same time, measured purely on the MQTT values, or still appearing to be fed into the house although the batteries are empty or have reached their discharge limit and the sun is no longer shining.

It would therefore be great if these 0 W per sensor could also be published via MQTT.

Faustilein, 2024-05-04:
I have the same problem with the Solar Input Power, Output Home Power, Pack Input Power, Output Pack Power sensors. Is there already a solution for this? Some of the values in the Home Assistant are not correct at all and you cannot keep any reasonable statistics. Thanks for a solution.

0Unzen, 2024-05-05:
solarInputPower updates in real time but packInputPower, outputPackPower and outputHomePower do not provide current values. As long as the values ​​don't flow properly you can't do anything.

Dani1802, 2024-05-19:
Still the same situation here, I'm hoping this issue will be fixed.
The Zendure app shows these values in real-time, but they are not published constantly over mqtt

JoB1l, 2024-05-29:
@z-master42 

I noticed that too and I have the same problem. Is there a way to solve this in the sensors themselves using the mqtt.yaml until Zendure gets it working again?

For example, if there is no new value for 5 minutes, it is set to 0?

z-master42, 2024-05-29:
Home Assistant has MQTT `expire_after`. I have already tried it, but nothing has changed. Maybe something is still missing 🤷‍♂️.

JoB1l, 2024-05-29:
@z-master42 

yes, I saw that by chance in your github project recently and added it to mine too. It does set the values ​​to 0 W if they are unknown (restart the Yaml configuration) but the values ​​are not reset overnight. I haven't found a solution yet as to how to do that.

Dani1802, 2024-05-29:
I also tried expires_after but that didn't work for me too.

Finally I decided to make two new template sensors, which set their state depending on the packstate.

    - name: "Battery In"
      unique_id: "battery_in_w"
      icon: mdi:flash
      unit_of_measurement: W
      state_class: measurement
      device_class: power
      state: >
        {% if states('sensor.packstate') in ['0', '2', 'unknown'] %}
          {{ 0 }}
        {% else %}
          {{ states('sensor.outputpackpower') }}
        {% endif %}

    - name: "Battery Out"
      unique_id: "battery_out_w"
      icon: mdi:flash
      unit_of_measurement: W
      state_class: measurement
      device_class: power
      state: >
        {% if states('sensor.packstate') in ['0', '1', 'unknown'] %}
          {{ 0 }}
        {% else %}
          {{ states('sensor.packinputpower') }}
        {% endif %}

0Unzen, 2024-06-08:
This should solve the problem with the missing 0 values, but the data is still incomplete because a new value only comes in via MQTT every minute.

JoB1l, 2024-06-08:
I have now temporarily solved the whole thing differently. I have created template sensors that calculate the charging power and discharging power for me.

"Solar input" - "inverter input"

so the value is either positive (charging) or negative (discharging). The negative value is converted back into a positive.

Unfortunately, something seems to have changed again with MQTT at Zendure. Now the "Solar Input" sensor is no longer real-time like it was all the time, but updates sporadically between 1-32 seconds :(

`  - sensor:
    # Calculating the battery power
      - name: "Battery power with negative values"
        unique_id: "battery_power"
        unit_of_measurement: 'W'
        device_class: "power"
        state_class: "measurement"
        state: >-
          {{ ((float(states('sensor.zendure_solarflow_hub_2000_solar_input_power')) * ( 0.95 ))) - float(states('sensor.hub_inverter_input_power')) | round(2) }}

  - sensor:
    # Battery charging power
      - name: "Battery charging power"
        unique_id: "battery_charging_power"
        state_class: "measurement"
        unit_of_measurement: 'W'
        device_class: "power"
        state: >
          {% if is_number(states('sensor.battery_power')) and states('sensor.battery_power') | float(0) < 0 %}
            {{(states ('0.0') | float(0)) |round(3) }}
          {% else %}
            {{(states('sensor.battery_power') | float(0))*1 | round (3) }}
          {% endif %}

  - sensor:
    # Battery discharge power
      - name: "Battery discharge power"
        unique_id: "battery_discharge_power"
        state_class: "measurement"
        unit_of_measurement: 'W'
        device_class: "power"
        state: >
          {% if is_number(states('sensor.battery_power')) and states('sensor.battery_power') | float(0) < 0 %}
            {{(states('sensor.battery_power') | float(0))*-1 | round (3) }}
          {% else %}
            {{(states ('0.0') | float(0)) |round(3) }}
          {% endif %}`

Raudi1, 2024-07-20:
This issue still persists and also effects the Hyper 2000. This completely messes up the Home Assistant and EVCC data.

z-master42, 2024-07-20:
> This issue still persists and also effects the Hyper 2000. This completely messes up the Home Assistant and EVCC data.

Is there already data from the Hyper via MQTT?

Raudi1, 2024-07-20:
Yes, since a few days ago.

jschroeter, 2024-09-10:
Same here - to do anything meaningful with it it's essential to get all data updated asap, which seems to be the case when using the app.

jschroeter, 2024-09-12:
My current ugly workaround to get up-to-date grid power input/output values from Hyper is the following:

- Shelly Plug S connected to Hyper
- Since the Plug S can't detect the direction, I use the `packState` sensor from MQTT which seems to be updated frequently. With this I can detect if Hyper is charging or providing power to the grid. For some reason `packState` (and also other values from MQTT) frequently are set to an empty string, therefore I wrapped it in `sensor.battery_state` to always keep the last valid value.
- Then I have two custom sensors which use the power value from Plug S depending on `sensor.battery_state`.

```
- platform: template
  sensors:
    battery_state:
      unique_id: "sensor.battery_state"
      friendly_name: "Battery state"
      value_template: >
        {% set hasValue = has_value('sensor.packstate') and states('sensor.packstate') != '' %}
        {{ states('sensor.packstate') | int if hasValue else this.state | default(0, true) }}

    battery_power_in:
      unique_id: "sensor.battery_power_in"
      friendly_name: "Battery in"
      unit_of_measurement: "W"
      value_template: >-
        {% if is_state('sensor.battery_state', '1') %}
          {{ states('sensor.shellyplusplugs_switch_0_power')|float(0) }}
        {% else %}
          {{ 0 }}
        {% endif %}

    battery_power_out:
      unique_id: "sensor.battery_power_out"
      friendly_name: "Battery out"
      unit_of_measurement: "W"
      value_template: >-
        {% if is_state('sensor.battery_state', '2') %}
          {{ states('sensor.shellyplusplugs_switch_0_power')|float(0) }}
        {% else %}
          {{ 0 }}
        {% endif %}
```

Likely I'll give https://github.com/reinhard-brandstaedter/solarflow-control a try soon as this setup is not really satisfying.

z-master42, 2025-01-20:
@Zendure Will anything else happen here?