Zendure/developer-device-data-report #50: No data in MQTT

lbouriez, 2024-05-29:
Hello, I had to change my SuperBase for the thrid time. Now MQTT doesnt report anything. I can properly login but there are no data:
[image]
I did send through postman the info but the appKey and secret are still the same.
Any idea ?

MimbaMonkeyHouse, 2024-06-13:
Time to fix this?

planbnet, 2024-06-28:
Same here `mosquitto_sub -t '#' -v -h mqtt.zen-iot.com -u XXXXXX-P XXXXXXXXXXXXXXXX` does not yield any messages for days

JoB1l, 2024-07-11:
is the problem still there? since yesterday afternoon 3pm I have sporadically no more data via mqtt

lbouriez, 2024-07-11:
I still have the issue, never been able to get back the data. I think zendure gave up on this...

Astromechdroide, 2024-07-13:
> is the problem still there? since yesterday afternoon 3pm I have sporadically no more data via mqtt

I also have this problem

jgrieger, 2024-07-16:
Me, too.

@Zendure Please respond to this. Will also open a support ticket directly.

I am connected to the EU server.

lbouriez, 2024-07-19:
Did someone tried to open a ticket to zendure, seems they are abandoning the product 😔