Zendure/developer-device-data-report #51: no data in MQTT since June 16

Be4fun, 2024-06-17:
No data being published in mqtt. Working until June 16th around 11PM CET
API registration is "successful operations"

pascalbourges, 2024-06-17:
hi, same issue from 17 june 4 hour AM french timezone

ChristophA, 2024-06-17:
Same here since 4 am.

Astromechdroide, 2024-06-17:
We also have no connection here in Europe.
{"code":400,"success":false,"data":{},"msg":"Please bind the device first"}* Connection #0 to host app.zendure.tech left intact

ge8020, 2024-06-17:
Can confirm that. Austria-Europe
No Data in MQTT Explorer, but App & Shelly Connection are working fine

H0d0rKF, 2024-06-17:
Same Problem Germany Europe

jeff1975, 2024-06-17:
Seit gestern kein Mqtt von Zendure Europa

Stievie72, 2024-06-17:
Was ist das für ein Service... da denkste zahlste bisschen mehr und dann kümmern die sich und dann wird es nicht einmal Kommentiert. Das nenne ich Kundenservice.

programmieramt, 2024-06-18:
> Was ist das für ein Service... da denkste zahlste bisschen mehr und dann kümmern die sich und dann wird es nicht einmal Kommentiert. Das nenne ich Kundenservice.

Wenn ich das richtig verstehe wird hier eine Dev-Schnittstelle genutzt. Die Kaufentscheidung anhand der Verfügbarkeit einer Dev-Api zu treffen ist wohl nicht ideal.
Die App funktioniert ja die ganze Zeit. Morgen beginnt soweit ich weiß die Intersolar und vermutlich muss da noch mal was fertig gemacht werden bevor die neuen Geräte gezeigt werden.

Stievie72, 2024-06-18:
> Wenn ich das richtig verstehe wird hier eine Dev-Schnittstelle genutzt. Die Kaufentscheidung anhand der Verfügbarkeit einer Dev-Api zu treffen ist wohl nicht ideal. Die App funktioniert ja die ganze [...]

Na dir Kaufentscheidung beinhaltet meines Erachtens nsch auch dir zur Verfügung gestellten Schnittstellen. Auch gerade wenn man solche Anlagen in eine smarte Hausinstallation integrieren möchte. 
Aber gut, mir geht es ja auch hauptsächlich um die Reaktion auf Kundenanfragen und Meldungen von Problemen. Da sollte es ein mindestens sein, solche Anfragen nicht einfach unkommentiert stehen zu lassen. Selbst wenn die Antwort nur beinhaltet, daß man das Problem sichtet, sobald die Zeit es zulässt. Aber 0 Reaktion setzt halt kein gutes Licht auf den Support. Würde ich so mit meinen Kunden umgehen,  muss ich mich nicht wundern, wenn der momentane Rekordumsatz irgendwann schwindet... aber gut, andere Baustelle ;)

H0d0rKF, 2024-06-18:
Bei mir kommen wieder Daten an.

python1185, 2024-06-18:
Seit 0800 kommen wieder Daten an.

Stievie72, 2024-06-18:
Hier auch ;)

jeff1975, 2024-06-18:
Danke Daten Ok

Be4fun, 2024-06-18:
Switched to global dev Server and Hope the information on changes will improve in the Future

closed 2024-06-18 by Be4fun

jaegerschnitzel, 2024-06-18:
Es wurde ja vor ca. einem Jahr gesagt, dass die Daten zukünftig auch lokal abgegriffen werden können. Gibts dazu Neuigkeiten?