Zendure/developer-device-data-report #52: Add support for Satellite CT and Hyper 2000

Troubadix, 2024-06-24:
Please add support for Satellite CT smart meter and Hyper 2000 battery/inverter

michije, 2024-07-04:
I received my Hyper2000 yesterday and was very disappointed by possibilities the app, so I looked into doing things myself. It's a shame that Zendure throws completly blocked off devices on the market, while others (ecoflow) offer broad API/MQTT support. I won't be keeping the Hyper if the app stays the same and no API support will be offered.

carfnann, 2024-07-04:
Hyper 2000 should definitely be supported by this API !

michije, 2024-07-04:
> Hyper 2000 should definitely be supported by this API !

Have you tried yourself? My account is set to "global", installed the Hyper 2000 with this account, global API returns error 400 (same for EU endpoint but this is expected)

carfnann, 2024-07-04:
yes I tried on both endpoint... I do have a success but it's because I do have a hub 2000 on my account. but in MQTT no data for the hyper

oliverritzmann, 2024-07-06:
Have the same, can connect to mqtt.zen-iot.com but there are no datas available for the Hyper2000. With a HUB1200 it was flawless, no chance with the Hyper. :(

towe75, 2024-07-11:
They added ACE and Hyper specific fields to the Readme a few days ago:  https://github.com/Zendure/developer-device-data-report/commit/f24cf68032bf9cdc9d367bee07c991d5bf7e1130 

Maybe it will be supported soon. Does it already work for somebody here?

michije, 2024-07-15:
I finally managed to receive MQTT data from the Hyper: the censored text is the appKey, for password use the secret. It's important to setup the topic in the format of my second screenshot: [appkey]/# QOS set to 1
[image]
[image]

carfnann, 2024-07-15:
Indeed, Hyper is now available as the hub 2000.
thanks for the news

z-master42, 2024-07-21:
> I finally managed to receive MQTT data from the Hyper: the censored text is the appKey, for password use the secret. It's important to setup the topic in the format of my second screenshot: [appkey]/# QOS set to 1 ![1](https://private-user-images.githubusercontent.com/78039822/348655797-609ca8d9-f051-4a1e-86c7-667ee032149e.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE1NjgyNjYsIm5iZiI6MTcyMTU2Nzk2NiwicGF0aCI6Ii83ODAzOTgyMi8zNDg2NTU3OTctNjA5Y2E4ZDktZjA1MS00YTFlLTg2YzctNjY3ZWUwMzIxNDllLkpQRz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzIxVDEzMTkyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZjMjcyNjI0OTM4NzdlMDRkZDY3ZTA0ZjkwMmRiMTdkMmU3YTU0YWYwNWVlYTMzZjcxN2RhM2RlZTJhNjQzYTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.AibEqH0j4VQo2rGjwqMJZpRhCw8NmgoOp6s86kq8pQA) ![2](https://private-user-images.githubusercontent.com/78039822/348655800-d70c66fa-8b95-4bcb-b2bd-92ad82122ac4.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE1NjgyNjYsIm5iZiI6MTcyMTU2Nzk2NiwicGF0aCI6Ii83ODAzOTgyMi8zNDg2NTU4MDAtZDcwYzY2ZmEtOGI5NS00YmNiLWIyYmQtOTJhZDgyMTIyYWM0LkpQRz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzIxVDEzMTkyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI0ZWEwZWNmZThlNmY5MTdkNTY4MDUwNzBjZjM0YmMzYjM5ZWVlYzkzM2ViMmE3ZDFmNmU1YjQwMTIxMGI2MWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.sWdLjxUr_b84zQ7wDcA--OpeAaJepPLF1VOJFGRG5H0)

Yes, the path is the same as before for the Hub 1200 and the Hub 2000.

jazz-o-mat, 2024-09-30:
Pls add support for Satellite CT. 
I need the information about the total power consumption of the house via mqtt.