Zendure/developer-device-data-report #53: Documentation for Hyper and ACE is incomplete

ppitkin, 2024-07-12:
The newly added valued on the description page appears to be incomplete for the Hyper.

When monitoring the values that are coming in the actual MQTT feed there are many fields contained there that are not listed.

Please provide a full ist of the parameters.

michije, 2024-07-13:
Hi @ppitkin , so you are already getting actual values from your Hyper? You seem to be the first one to have managed this, see #52

ppitkin, 2024-07-15:
Yes, I'm getting values on the MQTT feed,

z-master42, 2024-07-21:
For which values are entries still missing?

ulope, 2024-11-05:
Some are mentioned in #38

z-master42, 2024-11-05:
So ist ist only `acMode` ?

z-master42, 2024-12-10:
@Zendure can you please explain in detail what these fields represent?

Field | Description | device_class
-- | -- | --
gridInputPower | grid input power | sensor
acOutputPower | ac output power(Hyper 2000) | sensor
dcOutputPower | dc output power(Ace 1500) | sensor
acSwitch | ac switch | switch
dcSwitch | dc switch(Ace 1500) | switch

Furthermore what does acMode represent and what mode do the values stand for?