Zendure/developer-device-data-report #54: Control Hyper 2000 Charge / Discharge via MQTT

Soon5, 2024-07-18:
Hello,

I think about purchasing a Hyper 2000. I will be used as a AC coupled storage, therefore no solar panels connected. I want to use the internal charger to charge via AC and obviously discharge into the grid.

Is it possible to control both the charge / discharge rate via MQTT? My assumption is that setting outputLimit to 0 will stop discharging and gridInputPower to change the AC charge rate.

Is there someone who tried this? Does this work or is the API not yet capable to do this?

Regards

timofischer81, 2024-07-19:
I am also thinking about buying a Hyper 2000 and had exactly the same question wether charging/discharging can be controlled via API. This would be so cool. Anybody who already owns one has tested this?

Raudi1, 2024-07-20:
As far as I know it's currently not possible to control Zendure devices through MQTT. But the built in Shelly integration works great for using the Hyper 2000 as an AC storage. The only stupid thing is it tries to keep a 150W buffer while charging to prolong the life of the relay. But this doesn't help in any way because it will just turn off the relay when you hit below 150W going into the grid and turn back on when 150W is reached again. So it doesn't work as a buffer at all, doesn't prolong the life of the relay and wastes energy going into the grid.
A correct implementation would just try to lower the charging power to hit as close to 0W as possible. And only if the charging power of the zendure goes below a certain threshold, it should turn off. This would actually prolong the life of the relay and waste less energy.

Soon5, 2024-07-20:
The problem is that I have an environment here where the simple 3EM Integration wont work. Anyway, thanks for the answer.

iroc-fabi, 2024-07-24:
Ich möchte den Hyper Aufsatz auch als Speicher für meine große PV-Anlage nutzen.
Ich könnte die Werte vom SMA Energymeter aus iobroker oder HomeAssistent an den Hyper senden. 
Wird das per mqtt möglich sein?

michije, 2024-07-31:
> Ich möchte den Hyper Aufsatz auch als Speicher für meine große PV-Anlage nutzen. Ich könnte die Werte vom SMA Energymeter aus iobroker oder HomeAssistent an den Hyper senden. Wird das per mqtt möglich sein?

> The problem is that I have an environment here where the simple 3EM Integration wont work. Anyway, thanks for the answer.

> As far as I know it's currently not possible to control Zendure devices through MQTT. But the built in Shelly integration works great for using the Hyper 2000 as an AC storage. The only stupid thing [...]

While it's true that Zendure does not Support writing values via MQTT, there is a working and actively updated ioBroker Adapter which can update the output: https://github.com/nograx/ioBroker.zendure-solarflow
@iroc-fabi wenn du also den aktuellen Strombedarf vom Energymeter im ioBroker hast, kannst du diese Werte via ioBroker Skript an den Hyper schicken. Wichtig dabei: Einen zweiten Zendure User erstellen und dem das Gerät freigeben und dessen Logindaten in ioBroker nutzen, denn ansonsten wird man jedes mal aus der Handyapp ausgeloggt.

apunkt, 2024-09-24:
I also vote for local control via MQTT to bypass the cloud and also to enable complex controller requirements.
Shelly integration is nice to have, but tying these functions to certain vendors only increases complexity in local installations!
My wallbox wants to be controlled by vendorX meter.
My heatpump wants to be controlled by ventorY meter.
My Hyper2K wants to be controlled by Shelly meter.
My solarroof wants to be controlled by grid provider certified meter.

I refused all this, because all of a sudden you've got a box full of meters metering the same energy flow.
I have smart meter, that publishes the information to my local MQTT broker where all my devices can subscribe.
[image]

jebos, 2024-10-12:
There is an update that the shelly can now talk to hyper without any cloud. Does this mean that lokal mqtt Communication is now possible?

For me, not having this, is a deal breaker. I refuse to install multiple smart meters... i have all the information and just need the ability to control input and output programmatically.

MaisonGF, 2024-10-24:
iobroker works well, i think we need a local controller with the bt manager on github to avoid the “idle” issue (sometimes the battery stay on idle whatever mode we select).

@Zendure this is how you get your product an anti consumer aura.

DimitriGilbert, 2024-11-23:
anything other than local MQTT is NOT smart !
A little return on experience :
There's been a big storm in my area, power got knocked out for 12h, but the network came back 48h later which meant that my 4 solar panels and 2KWh battery were absolutely useless...
I understand you need grid power for the micro inverter to run, It does NOT need internet connection ! Neither for the appliance itself, nor the phone to control it from BT...

Mediacj, 2025-01-03:
Yes, I would also buy a Hyper 2000. It meets all my needs, except for the ability to control it via MQTT.

Nicolas-nwb, 2025-04-08:
The Hyper 2000 no longer seems to be on the market, and we’re still seeing no sign of MQTT support for charging or discharging. Honestly, I don’t think it will ever happen… This lack of technical roadmap says a lot about what we can expect from future Zendure products.

stefanrybacki, 2025-04-08:
I can confirm local MQTT allows charging and grid feed in. I am using it with home assistant automation to use non shelly smart meter.

jebos, 2025-04-08:
> I can confirm local MQTT allows charging and grid feed in. I am using it with home assistant automation to use non shelly smart meter.

Local? So no cloud? How did you switch the mqtt the hyper listening to?

stefanrybacki, 2025-04-08:
@jebos google for solarflow-bt-manager to find instructions.

DaPeace, 2025-04-08:
@stefanrybacki is this only possible via local cloud or is the local mqtt-connection the same as the one to zendures mqtt? I have a hyper 2000 + 4 Batteries here and tried the whole evening to switch the ac-power and between ac and dc. Could you give me an mqtt-endpoint and an example what to send to switch those 2 values? Problem is, i got the stuff today and im fairly new to that..

stefanrybacki, 2025-04-08:
@DaPeace I actually don't know the cloud mqtt provided by zendure. I directly switched to local only. Judging by the issues raised here I'd assume control via MQTT is only possible via local (no cloud) mqtt. The mqtt endpoint for controlling the hub is `iot/X/Y/properties/write`, with payload `{"properties": {"Z": V}}`, where X is the device identifier, Y the deviceid, Z the property to change to some value V, where Z is `inputLimit`, `outputLimit` and `acMode` (there is more, but for ac in and out those are enough). `inputLimit` and `outputLimit` are in Watts while acMode is 1 or 2, where 1 means ac-in and 2 ac-out.

Raudi1, 2025-04-08:
Control isn't possible via the cloud MQTT. But there's https://github.com/FireSon/Zendure-HA and https://github.com/nograx/ioBroker.zendure-solarflow/ which allow control as well. The HA integration even has a local control mode now in the latest prerelease by emulating a Shelly pro 3em.

thierrywehr, 2025-04-08:
fireson is connected to zendure cloud
no local handling

jgrieger, 2025-04-08:
> Control isn't possible via the cloud MQTT. But there's https://github.com/FireSon/Zendure-HA and https://github.com/nograx/ioBroker.zendure-solarflow/ which allow control as well. The HA integration even has a local control mode now in the latest prerelease by emulating a Shelly pro 3em.

That is not entirely true. If you take a look at FireSon's or reinhard-brandstaedter's repositories, you actually can control your e.g. Hub 2000 by simulating that your app/program is the Zendure app, using a mixture of Zendure app-specific credentials and your account-specific ones. No one knows how long this works, but it is a way. Based on the aforementioned inspirations I created tooling for our home automation.

Raudi1, 2025-04-09:
> fireson is connected to zendure cloud
> no local handling
> 
> Le mar. 8 avr. 2025, 21:29, Vrabetz ***@***.***> a écrit :
> […](#)
Look at the pre release. It currently does require a cloud connection as well, but the control of the power is local.

thierrywehr, 2025-04-09:
iot/<type>/<ID>/function/invoke{"arguments":[{"autoModelProgram":2,"autoModelValue":1245,"msgType":1,"autoModel":9}],"deviceKey":"<ID>","function":"deviceAutomation","messageId":87296534,"timestamp":1728220104}

iot/<type>/<ID>/function/invoke{"arguments":[{"autoModelProgram":0,"autoModel":0}],"deviceKey":"<ID>","function":"deviceAutomation","messageId":46736054,"timestamp":1728210016}

DaPeace, 2025-04-10:
That device drives me crazy. Since yesterday i try to set the device to discharge ( acMode = 2) and set the outputLimit. Setting the acMode is possible. The device reports this change. But setting the outputLimit is not possible. It stays at 0 all the time. Are there any hidden buttons i have to push to get it working?

DaPeace, 2025-04-18:
After solving the discharge perfectly, the hyper now charges itsself without any command and i cant figure out why.
During the day its charging with 600w until its full. Is there any datapoint that is causing this behavior?