Zendure/developer-device-data-report #57: Support for new Shelly 3EM 63T & Shelly 3EM 63W

steffenxx, 2025-03-07:
Shelly has launched 2 new 3EM's on the market, they are significantly smaller and can therefore be better installed in narrow power distributions. In terms of function, they correspond to the other Shelly 3EMs already supported by Zendure. 

The new Shelly's are not yet supported by Zendure. This means that they are not displayed in the Shelly account in the Zendure app. This means that Solar Flow (etc.) can no longer be operated in CT mode. 

Shortly before the solar season, it would be good if Zendure could provide support for the new Shelly via an update. I am sure that there are no major adjustments to be made, as it is only a new design and model name.

steffenxx, 2025-03-14:
no response ?

xabix, 2025-06-23:
Does thar explain why I see:
```
2025-06-20 15:27:36.952 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Adding device: ccc ShellyPro3EM 
2025-06-20 15:27:36.952 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Data: {'id': 193702, 'deviceKey': 'ccc', 'snNumber': 'ccc', 'name': 'Shelly 3EM', 'productId': 55, 'productKey': 'ggg', 'onlineFlag': '00601', 'productName': 'ShellyPro3EM ', 'wifiStatus': True, 'blueState': False, 'fourGStatus': False, 'isShareFlag': '00602', 'input': False, 'output': False, 'electricity': None, 'hasPack': False, 'restState': 0, 'upsMode': False, 'upgradeStatusDes': '', 'productType': 12, 'deviceType': 1, 'upgradeStatus': {}, 'bindId': 0, 'bindStatus': 0, 'batteryCode': '', 'packList': [], 'inputPower': 0, 'outputPower': 0, 'slowChargePower': 0, 'temperature': 0, 'temperatureUnit': 0, 'remainOutTime': 0, 'bindType': 0, 'seriesMode': 0, 'parallelMode': 0, 'networkType': 0, 'standard': 'c', 'isSwitch': False, 'url': 'https://control.shelly.cloud/images/device_images/SPEM-003CEBEU.png', 'remainOil': 0, 'genMode': 0, 'shortCode': '', 'phaseCheck': 0, 'pass': 0, 'clusterList': [], 'clusterNotice': False, 'nodeStatus': 0, 'clusterId': None, 'phase': None, 'reverseState': 0, 'meterId': 0, 'hemsId': 6537, 'totalPower': 0, 'keyFlag': 0}
2025-06-20 15:27:36.952 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Device ShellyPro3EM  is not supported!
```

steffenxx, 2025-06-23:
Shelly has launched new 3EM's with a different form factor (3EM 63T & 3EM 63W). They work exactly like the 3EM, but are still not supported by Zendure.