Zendure/developer-device-data-report #59: EU MQTT server does not send any data since today

Lineflyer, 2025-06-05:
I am using the MQTT URL  `mqtt-eu.zen-iot.com` since several months and it was stable ever since.

However today around 13:00 CEST it stopped sending new data. 
Connection can still be established but simply no data.

Once I change to the non-eu server `mqtt-eu.zen-iot.com` I get new data, which makes it pretty sure, that the problem is with the EU server not my devices.

bytesinmotion, 2025-06-06:
I have the same problem. I can't even connect with MQTT Explorer. I tried to get a new key that seems to work but no connection to mqtt-eu.

kaluxx, 2025-06-06:
same with me. appkey can be retrieved but no data.

tixoman, 2025-06-06:
Yes same Problem. Please fix it.

Lineflyer, 2025-06-06:
Still nothing today:

[image]

elcombato, 2025-06-07:
Since 18:00 CEST today I'm receiving data again.

Lineflyer, 2025-06-07:
Same here. Data again since 21:00 CEST